yaqwsx/jlcparts #5: Property filter ordering is very hard to use.

teegee, 2020-10-06:
Great tool, but the sorting of rows in the property filter make it incredibly annoying to select ranges of incremental values. See for example:

[image]

(The 13.6W entry should not be there, but should be between 13 and 14)

yaqwsx, 2020-10-06:
Hi, it is a known issue. The problem is that the source data contain everything as strings with no fixed format. Thus it is hard to interpret the values as a number so they can be sorted (they are sorted alphabetically now).

The code is ready for the implementation of heuristics to parse the data into meaningful data and to normalize them. Writing the heuristics is trivial, but extremely time-consuming, so I implemented only a single one. Feel free to submit a PR for one! It is simple, just extend [these functions](https://github.com/yaqwsx/jlcparts/blob/7e9b4fc7f57fcb60f6413f7414d93114c26bf751/jlcparts/datatables.py#L22-L40).

Note that the ultimate solution would be if LCSC provided the data in a normalized form. However, I am afraid it will not happen.

teegee, 2020-10-07:
I see what you mean.. I ran some of your travis steps manually to experiment a bit. The normalizeValue that you mention would still return a string that would fail to order properly in the same way, wouldn't it? You'd have to store it normalized numerically, taking into account SI prefixes etc..

The whole key normalizing thing becomes disgusting in some cases, e.g. looking at MOSFETs:

           {
                "FET Type": "N Channel",
                "Drain to Source Voltage(Vdss)": "60V",
                "Continuous Drain Current (Id) @ 25\u00b0C": "115mA",
                "Vgs(th) (Max) @ Id": "2.5V @ 250uA",
                "Rds On (Max) @ Id, Vgs": "7.5\u03a9 @ 500mA,10V",
                "Power Dissipation-Max (Ta=25\u00b0C)": "200mW",
                "Basic/Extended": "Basic",
                "Package": "SOT-23-3"
            }

(some sort of unicode fail on my part, but doesn't matter)

yaqwsx, 2020-10-07:
Currently, it returns a string. The plan is to return a tuple `(value, unit)` and properly handle it on the frontend. However, since I didn't plan to implement it in the first release, I let the functions to return a string to at least normalize the units.

teegee, 2020-10-07:
OK, so you would return 

  `"Continuous Drain Current (Id) @ 25\u00b0C": "115mA",`

as `(0.115, "A")`?

How about stuff like

  `"Vgs(th) (Max) @ Id": "2.5V @ 250uA",`
  `"Rds On (Max) @ Id, Vgs": "7.5Ω @ 500mA,10V",`

Add another tuple field? I.e. `(value, unit, qualifier)`

`(2.5, "V", "@ 250 uA")`

It seems that the key in `"Rds On (Max) @ Id, Vgs": "7.5Ω @ 500mA,10V",` could be parseable as a format for the value, but that sounds like it would be to much consistency to ask of the data entries..

yaqwsx, 2020-10-07:
I am thinking about creating a dictionary of values to be an attribute and allow sorting based on the entry of the dictionary.  I.e., `"Rds On (Max) @ Id, Vgs": "7.5Ω @ 500mA,10V"` would become:

```
{
	"format": "${Rds On (max)} @ ${Id}, ${Vgs}",
	"primary": "Rds On (max)",
	"values": {
		"Rds On (max)": [7.5, "Resistance"],
		"Id": [0.5, "Current"],
		"Vgs": [10, "Voltage"]
	}
}
```

Then the select box would display you the values according to the format string with an option to sort them based on a given key (possibly to filter them? However, I am afraid that filtering would yield pretty complex UI).

tenbaht, 2020-10-13:
Maybe it would be enough to normalize only a few, hand picked special properties where the difficult sorting hurts the most: Resistance for resistors, capacity (and maybe voltage) for capacitors, inductance (and maybe current) for inductors.
At least as a start.
Or instead of a full representation of complicated data like the Rds example, maybe keep it simple and just normalize the first numerical value found in a string. This value could be used internally as the sorting key, but print out the full string as before. This might not be always fool proof, but it would be a huge improvement and might be a lot easier to implement. Just scan for [0-9.-]+ and adjust accordingly if any of [npumkMG] is found right after that.

yaqwsx, 2020-10-17:
So I prepared an initial draft on the structured values. You can find it on a development version available at http://jlcparts-dev.honzamrazek.cz/

So far, I prepared only categories Transistor:MOSFET and Resistor:Chip Resistor - Surface mount. You can recognize the values that have been processed by having a meaningful name instead of "default" in the "Sort by" field. 

@teegee, @tenbaht, please, give it a try and report any problems you find. Once it works, I will write some basic processing for capacitors and inductors and merge to the upstream version.

teegee, 2020-10-18:
Looks great! Selecting ranges works fine now. I could narrow down my FET to a dozen candidates easily, which is much better than anything I managed to do on the jlcpcb/lcsc websites.. 

I was confused why obvious non-matching parts were included in the result, but I just had to make one of the columns 'required'. Those incorrect parts all had missing parameter values, such as item C171542. Maybe you could consider another tickbox to remove that sort of noise, since they can't be matched to anything anyway.

yaqwsx, 2020-10-18:
I sat down for a few hours and prepared preprocessing of attributes for the following categories:
- resistors
- diodes
- capacitors
- inductors
- transistors
- linear regulator

Could you, @teegee and @tenbaht look at it a briefly check that the values match with descriptions?

tenbaht, 2020-10-18:
On Sun, 18 Oct 2020 08:21:37 -0700
Jan Mrázek <notifications@github.com> wrote:

> I sat down for a few hours and prepared preprocessing of attributes
> for the following categories:
> - resistors
> - diodes
> - capacitors
> - inductors
> - transistors
> - linear regulator
> 
> Could you, @teegee and @tenbaht look at it a briefly check that the
> values match with descriptions?

Yes, beautiful! Sorting works like a charm! It is a joy to find new
parts now!

But I found two issues:

## canceled selections

1. Select resistors-chip resistor: The resistor parameter boxes appear.
2. Select one resistance value, eg. 100mR. The matching resistors are
listed.
3. Select MOSFET (without unselecting the resistor box first): The
MOSFET parameter boxes appear, together with the resistor parameters.
But the former value selection is gone and the full range of resistance
values is selected. The selection should be uneffected instead.
4. Select on value für Id.
5. Unselect resistors: The resistor parameter boxes are removed, good.
But the selection for Id gets lost and the full range is selected
again. It should be uneffected.

## Weird values in the Id list:

At the end of the MOSFED/Id list weird values like "6.9A, 6A" show up.
These values are totally unsorted and selecting one of them results in
a random looking collection of MOSFETs and an empty Id column.
Apparently, the value parser gets confused by some irregular
syntax.

Thank you for your work! This tool is so useful. I wish I have had it
for my last project. It took me almost a week to decide on the details
of the BOM for that one. I just did a fresh check for the key
components and after half an hour I already knew that I could have
saved half of the component cost. Whow.

Best regards,

  Michael

yaqwsx, 2020-10-18:
Thanks for the feedback.

- the first behavior "when you select a new category, all properties get selected" is intentional. When I tried to preserve the selected values between category selection, some users accidentally missed components - by not noticing that not all values are selected. I agreed with the beta testers, that the typical workflow is that you first select a category and then select properties. Selecting another category usually means that the user wants to start a new search, thus, selecting all properties makes sense. Personally, I am not a big fan of a button "start new search" or "restart the session". But I am open to discussion. However, if you would like to discuss it, please open a new issue as it is not related to this one.
 
- the "weird" values are intentional. There are MOSFETs with P and N transistors in a single package. By default, single MOSFETs have value "Id", the double MOSFETs have values "Id 1" and "Id 2". When you sort values by "Id", the properties without "Id" are listed at the end of the list. When you sort by "Id 1", the values for the double MOSFETs get sorted, and the single values are unsorted. If this is extremely confusing, I can change the double MOSFETs to have values "Id" and "Id 2" so they become sorted. But I kept them separate as usually you either look for a single MOSFET or a combined MOSFET and this helps you to select only relevant values.

BTW: If you like the tool(s) I make, you can consider supporting me on a one-time or regular basis.

yaqwsx, 2020-10-19:
The current version was applied upstream and is ready to use. If further values need normalization or change in the normalization, the users are expected to open a new issue.

closed 2020-10-19 by yaqwsx