yaqwsx/jlcparts #12: Tampermonkey to send found set of resistors to Resistor Paralleler to calculate a desired combination.

sleemanj, 2020-10-31:
Just something that you and other people might find useful, not really a FRQ or issue.

Years ago I wrote a parallel (and limited series) resistor calculator which used a given set of "stocked" resistor values rather than entire series.  This lends itself well to the JCB "basic" parts, since they are quite limited, using the paralleler you can calculate a set of resistors to meet a desired target value using only "basic" JLC parts.

The following Tampermonkey (GM probably fine too, only tested in Chrome), adds a button to do exactly that.  Screenshots below show how it works.

https://gist.github.com/sleemanj/b5e75f18d7292ab43e37b96e2a7ef905/raw/resistor-paralleler-for-jlcpcb-assembly-parts.js

[image]
[image]

sleemanj, 2020-10-31:
PS: If you feel motivated, feel free to rip the javascript in my paralleler to develop a paralleler right inside your tool.

yaqwsx, 2020-10-31:
Nice! I have to try it. I am thinking about where should it fit in the UI. Also, it would be nice to be able to compute resistor dividers in a similar way.

yaqwsx, 2021-02-04:
With the recently added history page, there is a suitable place for the UI: there are tabs. So I can imagine this feature could be implemented as a separate tab (showing the resistors category and its properties to filter).