yaqwsx/jlcparts #13: Display the low quantity fixed fee and/or attrition requirements.

sleemanj, 2020-11-02:
For low quantity of some parts, JLC assembly charges a fixed fee, these JLC minimum (eg 20 pieces for resistor is common) are not the same as LCSC minimum (eg 100 pieces for resistor is common).

[image]

Examples...

  * For qty≤20,charges a fixed fee $0.0380 https://jlcpcb.com/parts/componentSearch?searchTxt=0603WAF5101T5E 
  * For qty≤5,charges a fixed fee $3.8045   https://jlcpcb.com/parts/componentSearch?searchTxt=6TPE330ML 
  * For qty≤15,charges a fixed fee $0.2220 https://jlcpcb.com/parts/componentSearch?searchTxt=C20526

it would be good to have this JLC minimum shown in your tool with the pricing.

Also some parts have an attrition requirement, it would be useful to see this as well if the data is available, example below showing the attrition requirement for a part in the JLC quote process.

[image]

---
NB: Their explanation of attrition isn't great, it works like this... `Total_Parts_Charged = max( BOM + Attrition, MOQ ) * Price`

yaqwsx, 2020-11-02:
Unfortunately, JLCPCB does not provide this information in their [component table](https://jlcpcb.com/componentSearch/uploadComponentInfo) where we source the data from.

To acquire the data, we either have to scrape the page or make requests from the client to the JLC website. Fortunately, the data are available via a GET request on the following URL:

```
https://jlcpcb.com/shoppingCart/smtGood/getComponentDetail?componentCode=C23186
```

It gives you JSON with these relevant keys:
```
"leastNumber": 20,
"leastNumberPrice": 0.0380,
"lossNumber": 6,
```

I won't be able to implement the functionality in the next two weeks.

yaqwsx, 2020-11-07:
So the implementation is available at http://jlcparts-dev.honzamrazek.cz/. Could you test it and let me know if you spot any bugs? I will merge it to upstream if after I get your confirmation it works correctly.

sleemanj, 2020-11-07:
I'm not seeing it, where should I be looking?

[image]

yaqwsx, 2020-11-08:
Try it again - there was a problem with CORS on the deployed page I didn't notice.

sleemanj, 2020-11-08:
Works, but the calculation appears incorrect for the item I checked...

[image]

[image]

yaqwsx, 2020-11-08:
Ouch, the quantity is sometimes returned as string and then Javascript concatenates instead of adding /o\. It should be fixed now.

sleemanj, 2020-11-09:
Cool I checked some random ones and they seem ok to me :+1:

yaqwsx, 2020-11-09:
Ok, so I deployed it on the main page.

closed 2020-11-09 by yaqwsx