yaqwsx/jlcparts #22: Numerical order?

crosswick, 2020-12-20:
Hi, love this tool! Is it a known issue, or even difficult challenge that values are not in numerical order? Like this:

[image]

yaqwsx, 2020-12-20:
LCSC and JLC PCB provide all data as text (this is the `default` type). We have to specify how to interpret the data manually for each possible field. I specified the post-processing process for the most commonly used fields, however, I didn't manage to do it for all fields as it is a very time consuming and laborious process.

Adding new postprocessing isn't hard; you just have to specify a new property in [`attributes.py`](https://github.com/yaqwsx/jlcparts/blob/master/jlcparts/attributes.py) and specify the normalization procedure in [this function](https://github.com/yaqwsx/jlcparts/blob/master/jlcparts/datatables.py#L45). I cannot promise you to find time to implement this anytime soon, however, if you submit a PR I will be happy to merge it.

dougy83, 2024-03-11:
This issue is also occurring with common attributes, e.g. "Saturation current (isat)", which gets sorted as a string instead of a current, because the normalizeAttribute() function uses case-sensitive comparisons. Making that function case-insensitive seems to work.