yaqwsx/jlcparts #23: nothing works

phil123456, 2021-01-16:
I tried searching for 10k 0603 ,nothing shows up
I tried clicking everywhere, nothing happens

yaqwsx, 2021-01-16:
Hello phil. Thank you for your report, however, there a lot of information missing so I can reproduce the problem and fix it. 

Are you a first-time user? If so, you should follow the instructions in the yellow box that you should see:
[image]

If not, do you see a complete page? Did you try updating the library? Could you give me information about your browser and give me a screencast or at least a series of screenshots?

yaqwsx, 2021-01-25:
There has been no response; closing. If the problem persists, please, supply more information and reopen the issue.

closed 2021-01-25 by yaqwsx