yaqwsx/jlcparts #28: Running application locally

flaviut, 2021-02-13:
Hi, I'm trying to run the application locally to maybe do a few performance improvements, but I'm running into errors (`x is undefined`, "Cannot fetch categories index: Response is not JSON").

Can you write up a quick set of instructions on how you do local development on this? Thanks!

yaqwsx, 2021-02-13:
Could you tell me more about your setup? There are no special requirements; just everything has to be served from a web server and data tables have to be generated in a directory `data` placed in root. Just see the structure of the `gh-pages` branch: https://github.com/yaqwsx/jlcparts/tree/gh-pages

flaviut, 2021-02-14:
I'm just running a generic Linux system, I don't have any special setup or anything.

I figured it out and wrote some short instructions in #29.

I'm not looking to set up a proper web server or anything for this, just to do some local development.

PR #29 (2021-02-14): Add development instructions to README
Fixes #28
closed 2021-02-14 by yaqwsx