yaqwsx/jlcparts #30: Also Track Removed Parts

zoe1337, 2021-03-01:
It is great to have a catalog history, but it only shows new parts. Would it be possible to also track parts which are removed from the catalog?

I love this tool, thank you so much for it.

yaqwsx, 2021-03-02:
Do you have an example of a part that was removed from the catalog? Currently, I am under the impression, that JLC PCB only adds parts to the catalog and never removes them. They just put the stock to 0, which is really inconvenient as you can't tell if the component will ever be available. I talked to JLC PCB a few months ago about this issue and they said they plan a new component reservation system for the end of 2021.

zoe1337, 2021-03-02:
Everything from Worldsemi suddenly disappeared, and while it is probably some glitch on their side, I was *very surprised* when working on a project with parts I designed in 2 days ago.

yaqwsx, 2021-03-02:
Interesting. The stock was high a few days ago (I ordered some boars), so I assume it is a glitch on their side. I would contact JLC PCB support about this.

I will consider this functionality (it seems like a source of interesting data), but I can't give you any promises on when it will be done.

zoe1337, 2021-03-02:
of course, and thank you for working on this project!!

yaqwsx, 2023-01-11:
After a year, I reached the following conclusion:
- the database is so big that showing added and removed components is not straightforward as it requires care in GUI
- it also seems like something out of the scope of this project, however,
- the `cache.sqlite` tracks last time when the component was in stock so if you interested in some analysis, you can do so via SQL.

PS: The `cache.sqlite` can be obtained via:
```
wget -q https://yaqwsx.github.io/jlcparts/data/cache.zip
wget -q https://yaqwsx.github.io/jlcparts/data/cache.z01 || true
wget -q https://yaqwsx.github.io/jlcparts/data/cache.z02 || true
wget -q https://yaqwsx.github.io/jlcparts/data/cache.z03 || true
7z x cache.zip
```

closed 2023-01-11 by yaqwsx