yaqwsx/jlcparts #32: Tool shows part in-stock but it's not found at JLC

euler357, 2021-03-09:
Not sure what is going on but jlcparts finds a WS2812B-Mini (C114583) and shows 8343 of them in-stock but on JLC's site, the part can't be found by the part number nor the LCSC#

[image]
[image]

yaqwsx, 2021-03-10:
TLDR; currently, the component database si outdated (the last update was on Saturday), hence the "fail symbol" next to the repository name. I hope I will fix it in the next 48 hours.

During the weekend, JLC PCB broke their website (they dropped the library, then readded multiple components, then it was unavailable). The page started to work on Monday properly. This broke our incremental update mechanism. Unfortunately, that means I have to refetch practically the whole library again, which takes roughly 30+ hours. That's why the fix is not available yet.

yaqwsx, 2021-03-11:
The page should be fixed now. 

A side note: During fixing the page, I noticed that JLC PCB silently split their spreadsheet with components into two sheets, thus we missed about 20k of components recently. This is fixed, so the library now contains over 86k components.

euler357, 2021-03-11:
Thanks

-------
Chris K Cockrum
[https://accuforge.com](https://accuforge.com/)

[image]
I reject any and all terms and conditions associated with information sent to this email address. If you don't agree then don't send information to this email address.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2021-03-11 by yaqwsx