yaqwsx/jlcparts #33: The database query does not work with Firefox

yaqwsx, 2021-03-12:
Firefox gives a weird error on the search query:

[image]

This was working before, there were no relevant changes to the codebase. It seems that a large enough database kills Firefox.

noonien, 2021-04-02:
Also suffering from this, also on FF.

rluebess, 2021-04-04:
me 2. chrome and Edge works for me. FF never finish the request

yaqwsx, 2021-04-04:
I am not able to find the cause of the issue unfortunately and the Firefox error message is not helpful either. I at least implemented a note on Firefox that tells the user the page is broken in ac1c9a1.

If there is anyone who has at least some clue what's wrong, please, let me know! I am getting desperate.

zoe1337, 2021-04-06:
I'm probaby not helping, just wanted to share that deleting local storage does not solve the problem.

noonien, 2021-04-06:
I'd like to help debug.
Have you been able to verify if there's a specific commit that introduces this bug?
Perhaps the database is too large? Does it work with a smaller one?

yaqwsx, 2021-04-06:
There is no specific commit that caused that. So I suspect that the database is too large for Firefox to handle it properly. I opened an issue on Dexie: https://github.com/dfahlander/Dexie.js/issues/1264.

abrugsch, 2021-04-09:
FWIW, I just ran a database update and FF had no issues (apart from taking a while to complete)

martindevans, 2021-04-09:
I also just ran a successful database update (FF 87.0).

Not sure how fast it was in the past, but this time: `Library update took  103105 ms`

yaqwsx, 2021-04-09:
@abrugsch & @martindevans: Library update works fine, but the search query is broken. Is it working for you? 

It is normal for the database update to take up to two minutes; based on your internet speed & computer speed. IndexedDB is not the fastest database.

martindevans, 2021-04-09:
Searching does still seem to cause the same error.

abrugsch, 2021-04-09:
@yaqwsx if i do say a search for STM32 and only select Embedded Processors it's fine and returns results quickly. If I select all categories then it doesn't seem to return results. (Been waiting 5 mins now)

yaqwsx, 2021-04-09:
This observation confirms my suspicion, that queries with a large number of results kill Firefox. Based on your post, I added a message for the user when the query fails.

yaqwsx, 2021-04-09:
I was considering collecting the query in pages to overcome the Firefox limit, however, the efficient implementation (see [Dexie.js documentation](https://dexie.org/docs/Collection/Collection.offset()) is not compatible with filtering categories.

Implementing the pagination based on the documentation means that we have to implement category filters in JS, which will be probably very slow (every query then essentially returns the full DB contents - just like when doing a full-text search over all categories).

markasoftware, 2021-04-20:
Refreshing the database is quite slow, but there are no errors in the console and the application seems to work fine. I use Firefox 78.9 ESR on Debian Linux.

maksz42, 2022-01-19:
It seems that `components = await query.offset(1).toArray();` is working fine on FF for some reason, at least for me. Maybe that helps you figure it out.
And maybe it's possible to work with the database "on the fly" rather than converting it to an array? (I'm not a JS dev please don't laugh at me :P)

yaqwsx, 2022-01-19:
Thanks for the tip!

With the current size of the database, the whole project is fundamentally wrong. It was designed for a 50x smaller database than we have now. The problem is that IndexedDB is really limited in what it can do, so we have to often linearly walk through all the elements and filter them. I use `toArray` as it gives much better performance than `.each` - see [Dexie doc](https://dexie.org/docs/Collection/Collection.each()).

At this moment I am still not sure how to continue; I am still thinking about involving SQLite on the client-side, but I have no idea how much faster that will be. I know there was someone on Twitter thinking about making a server-based solution.

dzarda, 2022-05-21:
I'm willing to guess that using sql.js would solve the problem. Provided you get the model right (you will)

maksz42, 2022-05-21:
My theory is (I might be completely wrong) that there is some hard limit of ram and the browser goes out of memory.
https://github.com/maksz42/jlcparts/commit/f607adc4a11161ea289eb8e4281ef1937fde7afe
I almost got it working using `query.filter()` but it throws `Unhandled rejection: AbortError: Transaction aborted` and I can't figure out why. Without aborting the ongoing query it seems to be working fine on firefox but the performance is poor.

dzarda, 2022-05-21:
True, initiating the query allocates 3-4 G in seconds, then triggering the popup.

maksz42, 2022-05-21:
In my understanding, `query.filter()` filters components on the database side avoiding unnecessary memory allocation caused by `query.toArray()`.

dzarda, 2022-05-21:
The memory hog is actually observed even on Edge (7G) which gives up just as well (after a few more seconds)
[image]

dzarda, 2022-05-21:
Is this an issue of the search results yielding too many entries?

maksz42, 2022-05-21:
I guess so because when I select all categories without searching any text, the browser also dies.

yaqwsx, 2022-05-22:
Yes, it is a problem of very large search results since we convert them to an array (a limitation of IndexedDb). The implemented solution just doesn't scale well; we started with 15k components, now there are about 990k. We need to switch to another technology, however, I can't give you any timeline on the change.

closed 2022-06-01 by yaqwsx