yaqwsx/jlcparts #35: Prevent freeze after clicking "Clear search"

mairas, 2021-04-23:
Whenever you do a text search, all component categories and subcategories are selected. However, when the "Clear search" button is clicked, the categories remain selected, causing the page to freeze for a few seconds while the full catalog is processed.

To prevent this, could the category selections be cleared when the "Clear search" button is pressed?

yaqwsx, 2021-04-23:
Seems reasonable and handy. Implemented in ca19b49 (should be online in ~30 minutes). Note that the categories are cleared only when all categories were selected automatically. Otherwise, it would be annoying if the user lost selected categories by clearing the text.

mairas, 2021-04-23:
Ahhh, perfect now! Thank you!

closed 2021-04-23 by yaqwsx