yaqwsx/jlcparts #37: Consider moving to SQLite

yaqwsx, 2021-05-02:
Recently, we started to struggle with the performance of IndexedDb. [This blogpost](https://phiresky.github.io/blog/2021/hosting-sqlite-databases-on-github-pages/) mentions possible usage of SQLite. Let's have a look and find out if it could be a suitable solution.

PS: Definitely a layer of caching will be necessary to avoid refetching the database.

leo60228, 2021-11-11:
Chrome plans to deprecate WebSQL at some point in the future: https://www.chromestatus.com/feature/5684870116278272

leo60228, 2021-11-11:
...wait, this repo uses IndexedDB. Oops.

yaqwsx, 2021-11-11:
Huh? I am confused; this issue has nothing to do with WebSQL, nor the project uses it.

leo60228, 2021-11-11:
I specifically recalled that the project used WebSQL. I'm not sure why I thought that.

yaqwsx, 2022-02-16:
We should probably test how large is zipped SQLite DB is. There's a huge chance that it will be actually smaller than the JSON files we send (it is around 100 MB). If that's the case the update feature could just download the whole DB and store it in a cache and we don't have to hassle with sql.js-httpfs that has poor support and instead, we could use directly sql.js in a web worker. In this way, performance could be increased and we could solve the Firefox issue and also, make the update feature, much, much faster.

phiresky, 2023-11-09:
Just want to mention that SQLite has native wasm support now. They have a few "ideas" for how to do persistent storage https://sqlite.org/wasm/doc/trunk/persistence.md . You'd download the whole SQLite db (which should be pretty small when compressed as a whole), extract it to the origin private file system, then query it from there. I'm not sure how mature those are by now. If that doesn't work, absurd-sql or wa-sqlite are other options.

yaqwsx, 2023-11-11:
@phiresky : Native WASM SQLite and OPFS sounds exciting and something we should probably use. Thank you for letting me know.