yaqwsx/jlcparts #38: Allow filtering catalog updates to exclude NaN price or 0-stock items, or changed stock/price only?

dave2, 2021-05-13:
Since JLCPCB seem to be just copying the entire LCSC database over lately, there's a lot of noise in the catalog history page which is just never-stock parts. It might be nice to reword this to either exclude them, or list stock/price changes only instead of part presence in their list?

yaqwsx, 2021-05-19:
I think the history feature needs a rework - e.g., to include #30 and #26. However, I don't have a good idea how it should behave. I think that instead of listing new components we should list components "back in stock/out of stock" - that are components that went from 0 -> some stock and also components that went from some stock -> 0 stock and completely ignore the presence of a component in the databse.

What do you think? @zoe1337

dave2, 2021-05-19:
Firstly, great site :)

I've used the history a lot when they had fewer parts and were still planning to add things like connectors, and the history tab really suited that well, so it was useful when they had a smaller catalog of only parts they were reasonably-stocking. Now it's less useful because of the way they're loading their catalog (and not limiting scope to reasonably-stocking parts).

Thing is, right now if you have a 0 stock part you can't complete an order, which is a huge problem since you can't trigger any sort of backorder or "indication" of intent to buy. If a part goes from some to 0 stock it's a warning sign you maaaay need to redesign or substitute parts. Equally, I'm interested in a part that was 0 now has stock, that's useful to know esp if I have a few different parts I could use (eg, one design I have I could use any of the STM32F407/417/427/429/437/439).

Worryingly they do seem to have issues with their database, a couple of days ago the STM32F407VxT part disappeared from basic entirely, I expect it'll come back (this is not the first time), but I don't know if that will show up as just >0 stock or if it'd show up as a "new" item. >0 stock is just as good as "new" in that situation anyway.

I'd support a history table that was just stock level changes, either simplistically (0/not-0) or just any change in stock count.