yaqwsx/jlcparts #39: Part images not loading - API change?

neg2led, 2021-08-09:
Hi, I've noticed that recently all the part images are starting to fail to load; this appears to be linked to LCSC's recent website revamp, the content you scrape from the webpage to retrieve the images etc. has changed and the particular section with various image sizes is no longer present, so we get "no image found" for every part 😢 

I started a vague attempt at fixing this up myself;  it looks like you're jumping through a lot of hoops to find the part's product page, scrape the page for a script tag, and process out the contents of the script tag to extract attributes and image URLs? 

I'm not quite sure what would need changing to deal with the change in image URL presentation, but there _is_ one thing I've found that's probably helpful. The scraping doesn't seem to be necessary anymore, with the new site there's a web API you can call at `https://wwwapi.lcsc.com/v1/products/detail?product_code=<LCSC C number here>` which passes back a JSON response with all the part details; e.g. for C1525 (0.1uF 0402 basic capacitor) you get:

```
{
  "productId": 1877,
  "productCode": "C1525",
  "productModel": "CL05B104KO5NNNC",
  "parentCatalogId": 312,
  "parentCatalogName": "Capacitors",
  "catalogId": 313,
  "catalogName": "Multilayer Ceramic Capacitors MLCC - SMD/SMT",
  "brandId": 254,
  "brandNameEn": "Samsung Electro-Mechanics",
  "encapStandard": "0402",
  "split": 100,
  "productUnit": "Piece",
  "minPacketUnit": "Reel",
  "minBuyNumber": 100,
  "maxBuyNumber": -1,
  "minPacketNumber": 10000,
  "isEnvironment": true,
  "allHotLevel": null,
  "isHot": true,
  "isPreSale": false,
  "isReel": true,
  "reelPrice": 3,
  "stockNumber": 6888200,
  "stockSz": 4332700,
  "stockJs": 2555500,
  "productPriceList": [
    {
      "ladder": 100,
      "usdPrice": 0.0013,
      "currencyPrice": 0.0013,
      "currencySymbol": "US$",
      "discountRate": null
    },
    {
      "ladder": 1000,
      "usdPrice": 0.001,
      "currencyPrice": 0.001,
      "currencySymbol": "US$",
      "discountRate": null
    },
    {
      "ladder": 10000,
      "usdPrice": 0.0009,
      "currencyPrice": 0.0009,
      "currencySymbol": "US$",
      "discountRate": null
    },
    {
      "ladder": 50000,
      "usdPrice": 0.0008,
      "currencyPrice": 0.0008,
      "currencySymbol": "US$",
      "discountRate": null
    }
  ],
  "productImages": [
    "https://assets.lcsc.com/images/lcsc/900x900/20180914_Samsung-Electro-Mechanics-CL05B104KO5NNNC_C1525_front.jpg",
    "https://assets.lcsc.com/images/lcsc/900x900/20180914_Samsung-Electro-Mechanics-CL05B104KO5NNNC_C1525_back.jpg",
    "https://assets.lcsc.com/images/lcsc/900x900/20180914_Samsung-Electro-Mechanics-CL05B104KO5NNNC_C1525_blank.jpg"
  ],
  "pdfUrl": "https://datasheet.lcsc.com/lcsc/1810191219_Samsung-Electro-Mechanics-CL05B104KO5NNNC_C1525.pdf",
  "productDescEn": null,
  "productIntroEn": "X7R 100nF ±10% 16V 0402 Multilayer Ceramic Capacitors MLCC - SMD/SMT ROHS",
  "paramVOList": [
    {
      "paramCode": "param_10954",
      "paramName": "温度系数",
      "paramNameEn": "Temperature Coefficient",
      "paramValueEn": "X7R",
      "isMain": true,
      "paramType": null,
      "paramDetailValue": "X7R",
      "paramUnit": null,
      "paramDetailValueForSearch": -1,
      "paramSearchType": null
    },
    {
      "paramCode": "param_10951_n",
      "paramName": "容值",
      "paramNameEn": "Capacitance",
      "paramValueEn": "100nF",
      "isMain": true,
      "paramType": "param_detail",
      "paramDetailValue": null,
      "paramUnit": null,
      "paramDetailValueForSearch": 100000,
      "paramSearchType": "equal"
    },
    {
      "paramCode": "param_10952_s",
      "paramName": "精度",
      "paramNameEn": "Tolerance",
      "paramValueEn": "±10%",
      "isMain": true,
      "paramType": "param_detail",
      "paramDetailValue": "±10%",
      "paramUnit": null,
      "paramDetailValueForSearch": null,
      "paramSearchType": null
    },
    {
      "paramCode": "param_10953_n",
      "paramName": "额定电压",
      "paramNameEn": "Voltage Rated",
      "paramValueEn": "16V",
      "isMain": true,
      "paramType": "param_detail",
      "paramDetailValue": null,
      "paramUnit": null,
      "paramDetailValueForSearch": 16,
      "paramSearchType": "able"
    }
  ],
  "productArrange": "Tape & Reel (TR)",
  "productWeight": 0.013,
  "foreignWeight": 0.013,
  "isForeignOnsale": true,
  "isAutoOffsale": false,
  "isHasBattery": false,
  "title": "Samsung Electro-Mechanics CL05B104KO5NNNC",
  "weight": 0.013,
  "brandCategory": 2,
  "videos": [
    {
      "url": "https://assets.lcsc.com/attachment/video/video_1b65dfd2cf7d46a08fa134d0a3a69e11.mp4",
      "cover": "https://assets.lcsc.com/attachment/video/video_8d167f6069fb4dda8f0c7f52583f3e84.jpg",
      "detail": "Let's see how to pick Multilayer Ceramic Capacitors at LCSC Electronics."
    },
    {
      "url": "https://assets.lcsc.com/attachment/video/video_74d949def8c245b6ae820630b61345cb.mp4",
      "cover": "https://assets.lcsc.com/attachment/video/video_3eced888029042fda97994a72903a92e.jpg",
      "detail": "See how to pick MLCC-SMD in LCSC"
    },
    {
      "url": "https://assets.lcsc.com/attachment/video/video_1376295b190a485793576a4ce0511d21.mp4",
      "cover": "https://assets.lcsc.com/attachment/video/video_f14d7328866d4e2f9cda92fa1119faa7.jpg",
      "detail": "How do LCSC staffs pick electrolytic capacitors?"
    }
  ],
  "recommend": [
    "Snipped this bit out for brevity since it's not needed."
  ]
}
```
(I ran this through `jq` for formatting)

I would be quite happy to rework things to use this API path to retrieve part info, but I haven't been able to work out enough of the program flow / how the objects that are being passed around are supposed to look to know what to change...

yaqwsx, 2021-08-17:
Thank you for letting me know and pointing me to the API. I was scraping the pages after the change in a hard way.

I just updated the system, so the images should gradually appear (we have to refetch the whole database which can take several days).

neg2led, 2021-08-18:
No problem, the whole scraping thing looked like it was a nightmare to set up & maintain, and I figured if they had a search API they probably had a "get product details" API - a little bit of digging through page sources later, I found one 😄 Thanks for the fix! 

I should mention, links through to the product pages (by clicking the C-number) weren't working for me either, but it seems like that probably had the same root cause; if that's still broken once parts are updated (it could take up to two weeks, from the look of things - at 10k/day with ~150-160k parts) I'll let you know.

Thanks again!

yaqwsx, 2021-08-22:
Both links and images should be working now.

neg2led, 2021-08-24:
All looks fine from here, thanks again! I'll go ahead and close this 😄

closed 2021-08-24 by neg2led