yaqwsx/jlcparts #40: Feature Request: Provide the ability to save/bookmark queries

brianredbeard, 2021-08-31:
## Feature Request

Provide the ability for users to save or "bookmark" a set of options within the UI making it easy to recall them.  

## Background

While searching for parts I'm often left going through a similar workflow after schematic capture:

1. Generate a BOM, sorting by component type, footprint, and value
2. Locate congruent parts within JLCPCB's currently available parts

This results in a lot of duplicated work:

1. First identify the category type a part lives in (e.g. `Multilayer Ceramic Capacitors MLCC - SMD/SMT`)
2. Filter on`Basic` part type
3. Add `Require on stock` parameter
4. Select the value (e.g. capacitance, resistance, etc) needed
5. Optionally restrict by footprint
6. Browse through available parts

Unfortunately this is challenging due to the density of information on the page.  As a single example:

[image]

It is common to have to repeat this workflow for all of the "jellybean" components:
- Capacitors
- Resistors
- Inductors
- LEDs

yaqwsx, 2021-09-01:
We could reflect all the parameters in URL and therefore, you could just bookmark them in your browser. Would you find this as a sufficient solution? In this way, you could also share your query with others (via URL).

yaqwsx, 2021-09-14:
I just looked at the possible implementation; all suitable libraries, unfortunately, use react hooks and unfortunately, this application is based on the older model, react classes. So this feature will require significant refactoring of the code.

closed 2026-05-25 by yaqwsx