yaqwsx/jlcparts #41: components are not showing

shiroichiheisen, 2021-09-16:
some components such as data converters are not showing when we select the category, for example, the analog switches & multiplexers category are not showing all the components that you have on the jlpcb website, it also happens that the website asks you to update the database but even updating it it still keeps asking again, i already deleted the cookies, I tested another browser and I even deleted the data from my browsers and tried again to update the database, but it still asks that it needs to be updated

shiroichiheisen, 2021-09-16:
I was looking through the jlcpcb website, and I ended up discovering that they also have this problem, if I search with the component code I find it, but if I go to the categories and try to find it there, I think there is some problems with them too

yaqwsx, 2021-09-17:
The problem you describe originates in the data table of components JLC PCB provides. There's unfortunately nothing I can do about it.

shiroichiheisen, 2021-09-17:
i tryed to read the csv from the table jlcpcb list for download, and i have the error on the archive, its says the table have 13 colunns but the title only have 12, there some way this its the problem?

shiroichiheisen, 2022-01-25:
was resolved, just closing the old issue

closed 2022-01-25 by shiroichiheisen