yaqwsx/jlcparts #50: Price computation ignores minimum quantity

eiffel31, 2022-02-15:
The feature with price computed for a specific quantity is really wonderful.

A case has to be fixed: the minimum quantity is not considered when computing the price.

See screen shots.
[image]
[image]
So, for any quantity below 100, the price will be $2.2536

yaqwsx, 2022-02-15:
This seems like the JLC PCB API has changed (again) and thus we don't extract the MOQ correctly. The MOQ is incorporated in the price calculation: https://github.com/yaqwsx/jlcparts/blob/c2750d85ed56368fc87e6418fddd951afb79db76/web/src/jlc.js#L77-L81

yaqwsx, 2022-02-15:
This seems like a bug on the JLC PCB API - the response we got for the component indeed shows that the MOQ is one: 
```
{
   "code":200,
   "data":{
      "assemblyMode":"manualWeld",
      "assemblyProcess":"THT",
      "attributes":null,
      "canPresaleNumber":145040,
      "componentBrandEn":"Xucheng Elec",
      "componentCode":"C52878",
      "componentDesignator":"U",
      "componentImageUrl":"https://assets.lcsc.com/images/lcsc/900x900/20190506_Xucheng-Elec-EST026_C52878_front.jpg",
      "componentLibraryType":"expand",
      "componentModelEn":"EST026",
      "componentName":"Xucheng Elec EST026",
      "componentPresaleType":"buy",
      "componentSource":"shop",
      "componentSpecificationEn":"Through Hole",
      "componentStatus":"yes",
      "componentTypeEn":null,
      "dataManualUrl":"https://datasheet.lcsc.com/lcsc/1810271721_Xucheng-Elec-EST026_C52878.pdf",
      "describe":"Quick connect terminal Plugin Cold-pressed?Terminals ROHS",
      "encapsulationNumber":5000,
      "encapsulationUnit":"PCS",
      "erpComponentName":"【压接】6.3mm*0.8厚 插片",
      "firstTypeNameEn":"Connectors",
      "initialPrice":0.0182,
      "jlcPrices":null,
      "lcscComponentId":53893,
      "lcscGoodsUrl":"https://lcsc.comhttps://lcsc.com/product-detail/Cold-pressed-Terminals_Xucheng-Elec-EST026_C52878.html/?href=jlc-SMT",
      "leastNumber":1,
      "leastNumberPrice":0.0256,
      "lossNumber":0,
      "matchedPartDetail":"Xucheng ElecQuick connect terminal Plugin Cold-pressed?Terminals ROHS",
      "minImage":"https://assets.lcsc.com/images/lcsc/96x96/20190506_Xucheng-Elec-EST026_C52878_front.jpg",
      "minPresaleNumber":0,
      "minPresalePrice":null,
      "minPurchaseNum":5000,
      "overseasStockCount":145740,
      "preMinPurchaseNum":null,
      "prices":[
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":999,
            "lcscComponentId":null,
            "productPrice":0.0256,
            "startNumber":100,
            "updateTime":null
         },
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":2999,
            "lcscComponentId":null,
            "productPrice":0.0213,
            "startNumber":1000,
            "updateTime":null
         },
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":9999,
            "lcscComponentId":null,
            "productPrice":0.0182,
            "startNumber":3000,
            "updateTime":null
         },
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":49999,
            "lcscComponentId":null,
            "productPrice":0.0156,
            "startNumber":10000,
            "updateTime":null
         },
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":99999,
            "lcscComponentId":null,
            "productPrice":0.0152,
            "startNumber":50000,
            "updateTime":null
         },
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":199999,
            "lcscComponentId":null,
            "productPrice":0.0149,
            "startNumber":100000,
            "updateTime":null
         },
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":399999,
            "lcscComponentId":null,
            "productPrice":0.0148,
            "startNumber":200000,
            "updateTime":null
         },
         {
            "componentCode":"C52878",
            "componentPriceAccessId":null,
            "componentPriceKeyId":null,
            "createTime":null,
            "deleted":false,
            "endNumber":-1,
            "lcscComponentId":null,
            "productPrice":0.0147,
            "startNumber":400000,
            "updateTime":null
         }
      ],
      "secondTypeNameEn":"Cold-pressed?Terminals",
      "stockCount":145740,
      "storyGoods":false,
      "warehouseCode":"sz",
      "warehouseName":"深圳"
   },
   "message":null
}
```

eiffel31, 2022-02-16:
Just an idea: as there is no price for quantity bellow 100 pcs, then we can consider that 100 pcs is the minimum quantity. Correct?