yaqwsx/jlcparts #53: [BUG] Price sorting problem

eiffel31, 2022-02-24:
How to reproduce:
Select "Power Management" -> "Monitors & reset circuit"
Order by stock:
[image]
The first one is MAX809S, with price $0.052

Now, order by price:
[image]
The lowest listed price is $0.054 

Unable to find MAX809S anymore (and other low cost references are missing).

BTW : the component matching query has a much lower count than the official web site for the same category.

yaqwsx, 2022-02-24:
Regarding the comment in BTW: JLC PCB has blocked all my update runners, so I am not able to provide updates for over a week. Therefore, the whole library is about 10 days old.

eiffel31, 2022-02-24:
The lock is not a good news: you provide a great service and help to find products to buy! Is there a way to discuss with JLCPCB to request for an unlocking exception?

yaqwsx, 2022-02-24:
I contacted them 5 days ago, so far, without response.

eiffel31, 2022-02-24:
Using the chat, they are highly responsive (except a few night hour of course).

eiffel31, 2022-03-04:
Still out of topic from initial bug report:
Happy to see that update is working again.
Do you know than on https://jlcpcb.com/parts there is a link "Download Table"? This way you quickly have all products in a single file.
This API problem would be better addressed in a specific issue.

yaqwsx, 2022-03-04:
The update is still not working (indicated by the red cross next to the repository name). We already use the table; we use LCSC API to download part attributes (e.g., values you can filter by). JLC PCB doesn't provide that data. And without that data this project is useless.

At the moment, where the future of the project is unclear, I will not provide any bug fixes (but I am willing to merge them if someone sends PR). If we get a reliable source of data, then I will address all the pending issues.

eiffel31, 2022-03-04:
This morning, the web page processed an update. The message is green and says that the database is up to-date.

When we go directly on https://yaqwsx.github.io/jlcparts/
we do not see any database problem. I guess you are referring to the red cross in the changelog.

yaqwsx, 2022-03-04:
There's no update. You can watch the updates at https://github.com/yaqwsx/jlcparts/actions. The last update was 17 days ago: https://github.com/yaqwsx/jlcparts/tree/gh-pages

yaqwsx, 2022-03-22:
Having the service running again I cannot reproduce this issue anymore. Could you confirm it?

yaqwsx, 2022-03-22:
Problem found - some components are missing prices and thus, they break sorting. Fixed in 931693a3098fa0c0692a5d8b0ef69f6f845f7fd8

closed 2022-03-22 by yaqwsx