yaqwsx/jlcparts #54: Allow Filtering of Hand-Soldered  Components

ahalekelly, 2022-03-03:
Thanks for the useful tool! Does the API expose which components require hand soldering? It would be great to be able to filter them out.

yaqwsx, 2022-03-22:
At the moment, JLC PCB doesn't provide this information in their spreadsheet. So I have no way of obtaining the information

raspico, 2023-07-17:
As an aside, REQUIRING hand soldering vs. ALLOWING it is obviously related to the package type. Maybe it would make sense to classify by package types: e.g SOP and DIL are easily solderable, QFN and TSSOP are harder,  and obviously BGA are almost impossible to hand solder.

yaqwsx, 2024-01-07:
@maksz42 Any changes you have an idea where to obtain this information on JLC PCB API since you poked into it recently?

maksz42, 2024-01-13:
@yaqwsx I found `https://jlcpcb.com/api/overseas-pcb-order/v1/shoppingCart/smtGood/getComponentDetail` which returns
```
"assemblyMode": "manualWeld"
"assemblyModeBatch": "manualWeld"
"assemblyProcess": "THT"
```
or
```
"assemblyMode": "smtWeld"
"assemblyModeBatch": "smtWeld"
"assemblyProcess": "SMT"
```
and some other interesting stuff.
Used on https://jlcpcb.com/parts/componentSearch?searchTxt=74hc595
A request is sent when a list item is expanded. It takes some LCSC's internal component code which is different from C12345 codes.
example:
```
https://jlcpcb.com/api/overseas-pcb-order/v1/shoppingCart/smtGood/getComponentDetail?componentLcscId=1359
```
[image]

Also, smd/tht can be somehow infered from LCSC search https://www.lcsc.com/search?q=74hc595
which has endpoint `https://wmsc.lcsc.com/wmsc/product/search/list`
```
minPacketUnit	"Reel"
productUnit	"Piece"
productArrange	"Tape & Reel (TR)"
```

yaqwsx, 2026-05-25:
This was implemented in b9ea2de394a448eb7748f76997c3c9f117afa9f3. It will, however, take a couple of days to fully update  the database (due to JLC PCB rate limits).

closed 2026-05-25 by yaqwsx