yaqwsx/jlcparts #55: [BUG] Quantity error: extra 0

eiffel31, 2022-03-04:
The quantity should be 1-9, 10-29, 30-99...
Unfortunately, it is displayed with an extra 0, giving 10-90, 100-290, 300-990...

It is not just a display formatting error, because price considered for 1000 pcs is calculated according to the wrong quantity value.

Another problem (may be related or not): the price is different from the price on the official website.

Example below shows product C2683482.

[image]
[image]

yaqwsx, 2022-03-04:
This is an inconsistency in data provided by JLC PCB obtained during the last update. Once we can update again, it should be working normally.

yaqwsx, 2022-03-22:
Once updated everything seems to work fine:

[image]

closed 2022-03-22 by yaqwsx