yaqwsx/jlcparts #56: [BUG] Computed price incoherent display

eiffel31, 2022-03-04:
Price for the requested quantity is displayed in the price column and in the detailed description. The values differ.

For product C20526, quantity 1000 selected. 2 different prices are displayed for this quantity.
[image]

Note: here, quantity range displayed are strange. Is it related to the minimum quantity 15? As displayed, there is no possibility to buy 140 pcs due to the gap between 135 and 150.

yaqwsx, 2022-03-04:
The price is computed from two sources:

- the price you see in the table comes from the data table provided by JLC PCB at https://jlcpcb.com/componentSearch/uploadComponentInfo and it is updated once a day (when we are not blocked by JLC PCB)
- the price you see in the detail is fetched from https://jlcpcb.com/shoppingCart/smtGood/getComponentDetail?componentLcscId={ID} every time you see it so it should be the most up-to-date price.

As JLC PCB provides poor data, I am afraid I cannot provide a better service. At the moment, it is possible that the data are just out-of-sync as the table operates on 14 days old database at the moment.

eiffel31, 2022-03-04:
I do understand your problems.
I also gave feedback to JLCPCB for some inconsistencies I observed today. I hope they will be able to fix them.

About the lock, where you able to use the contact as described here? https://support.jlcpcb.com/article/71-jlc-api-solution
I hope they will be able to white list your service.

yaqwsx, 2023-11-05:
This should no longer be a problem as we fetch price directly from JLC PCB API

closed 2023-11-05 by yaqwsx