yaqwsx/jlcparts #71: New API

EmerickH, 2022-07-16:
Hey,
I just wanted to point out that JLCPCB has released a new way to browse thru components available in your account:
https://smt.jlcpcb.com/smtPrivateLibrary?index=1
The new page seems to be working with a (possibly new?) API at:
https://smt.jlcpcb.com/overseasSmtComponentOrder/componentSearch

**Example request:**
```bash
curl -i -H "Content-Type: application/json" -X POST -d '{"currentPage":1,"pageSize":10,"componentSpecification":"","keyword":"","componentType":"","secondeSortName":"Light Emitting Diodes (led) ","componentLibraryType":"base","searchSource":"search","firstSortName":"Optocouplers & Leds & Infrared ","componentBrand":"","componentAttributes":[]}' https://smt.jlcpcb.com/overseasSmtComponentOrder/componentSearch/selectSmtComponentList
```

<details>
  <summary><b>Example answer</b> (click to develop)</summary>

 ```JSON
{
  "code": 200,
  "msg": "Successful",
  "data": {
    "componentPageInfo": {
      "endRow": 0,
      "hasNextPage": false,
      "hasPreviousPage": false,
      "isFirstPage": false,
      "isLastPage": false,
      "list": [
        {
          "canPresaleNumber": 217265,
          "componentBrandEn": "Hubei Kento Elec ",
          "componentCode": "C2296",
          "componentId": 2669,
          "componentImageUrl": "https://assets.lcsc.com/images/lcsc/900x900/20180914_Hubei-KENTO-Elec-17-21SUYC-TR8_C2296_front.jpg",
          "componentLibraryType": "base",
          "componentModelEn": "17-21SUYC/TR8",
          "componentName": "Hubei KENTO Elec 17-21SUYC/TR8",
          "componentPrices": [
            {
              "endNumber": 19980,
              "productPrice": 0.0084,
              "startNumber": 10000
            },
            {
              "endNumber": -20,
              "productPrice": 0.0079,
              "startNumber": 80000
            },
            {
              "endNumber": 1980,
              "productPrice": 0.0099,
              "startNumber": 600
            },
            {
              "endNumber": 79980,
              "productPrice": 0.008,
              "startNumber": 40000
            },
            {
              "endNumber": 180,
              "productPrice": 0.0137,
              "startNumber": 20
            },
            {
              "endNumber": 580,
              "productPrice": 0.0113,
              "startNumber": 200
            },
            {
              "endNumber": 9980,
              "productPrice": 0.0091,
              "startNumber": 2000
            },
            {
              "endNumber": 39980,
              "productPrice": 0.0081,
              "startNumber": 20000
            }
          ],
          "componentProductType": 0,
          "componentSource": "shop",
          "componentSpecificationEn": "LED_0805",
          "componentTypeEn": "LIGHT EMITTING DIODES (LED)",
          "dataManualUrl": "https://datasheet.lcsc.com/lcsc/1806151129_Hubei-KENTO-Elec-17-21SUYC-TR8_C2296.pdf",
          "describe": "YELLOW 592~594NM 0805 LIGHT EMITTING DIODES (LED) ROHS",
          "erpComponentName": "LED_灯_黄_Iv=95-113_λd=592-594",
          "firstSortAccessId": "f972da204e9147268b09e5e9b0c6292d",
          "lcscGoodsUrl": "https://lcsc.com/product-detail/Light-Emitting-Diodes-LED_Hubei-KENTO-Elec-17-21SUYC-TR8_C2296.html",
          "mergedComponentCode": null,
          "minImage": "https://assets.lcsc.com/images/lcsc/96x96/20180914_Hubei-KENTO-Elec-17-21SUYC-TR8_C2296_front.jpg",
          "secondSortAccessId": "ac4c3bf204dc486cb0c9c3fc16ef0c61",
          "stockCount": 249140,
          "urlSuffix": "Hubei_KentoElec-17_21SUYCTR8/C2296"
        },
        [ALL OTHER COMPONENTS....]
      ],
      "navigateFirstPage": 0,
      "navigateLastPage": 0,
      "navigatePages": 0,
      "navigatepageNums": null,
      "nextPage": 0,
      "pageNum": 1,
      "pageSize": 10,
      "pages": 1,
      "prePage": 0,
      "size": 0,
      "startRow": 0,
      "total": 5
    },
    "sortAndCountVoList": [
      {
        "childSortList": [
          {
            "childSortList": null,
            "componentCount": 5,
            "componentSortKeyId": null,
            "grade": null,
            "parentId": null,
            "sortImgUrl": null,
            "sortName": "Light Emitting Diodes (led) ",
            "sortUuid": null
          }
        ],
        "componentCount": 5,
        "componentSortKeyId": null,
        "grade": null,
        "parentId": null,
        "sortImgUrl": null,
        "sortName": "Optocouplers & Leds & Infrared ",
        "sortUuid": null
      }
    ]
  }
}
```  
</details>

I think it could be useful to fetch the components, and maybe an uncached version of this app could be cool? What do you think about it?

yaqwsx, 2022-07-16:
Thanks for pointing that out. I will look at it. However, JLC PCB still offers fewer details and parameters for components, so I still plan to maintain a custom database. Also, we normalize most of the attributes so they can be easily filtered. If we gave the database we would loose all the component attributes and also, we wouldn't be able to perform search.

EmerickH, 2022-07-16:
Well unlike the old API, it seems this one does allow to filter directly with characteristics, example with a request like this:
```JSON
{
  "currentPage": 1,
  "pageSize": 10,
  "componentSpecification": "0603",
  "keyword": "my search",
  "componentType": "",
  "componentLibraryType": "base",
  "searchSource": "search",
  "componentBrand": "",
  "componentAttributes": [
    {
      "Illumination Color": "Green"
    },
    {
      "Mounting Type": "Surface Mount"
    }
  ]
}
```

There is also a request to get what filters are available:
https://smt.jlcpcb.com/overseasSmtComponentOrder/componentSearch/filterComponentAttribute

Yeah being able to redefine categories and all in our custom database is not bad, but maybe to avoid it to be absolutely massive it could be possible to make it dynamical (only request a category when you actually need them) if we get it to be faster to download data that it is right now (with eventually an option to download the entire library). I feel like it's not very convenient right now to have to wait 20min (with a beefy CPU) to generate 3.3GB of database with 99% components I'm never going to search thru...

yaqwsx, 2022-07-16:
The database size and speed will improve drastically once I find time to implement #37.