yaqwsx/jlcparts #73: No stock number and missing components.

UniversalChaos, 2022-10-08:
Hi.

Fell in love with this tool, but unfortunately for the last 1-2 weeks it has been broken. There is no number to stock amount and components are missing from categories (eg. the HALL sensor category contains 2 components on this site, but pages on jlcpcb).

Sorry to bother.

yaqwsx, 2022-10-09:
This is intentional. As the page was designed for 20k components and not 1.2 million components, the technology used doesn't scale well (see #37). Therefore, the catalog was very, very slow. However, from the 1.2 million components, only about 200k are in stock, and I believe that more than 500k components were never in stock.

Therefore, in 487d1321dd2e53ec75c65ed1852c6143d4f79df2 I introduced a change that ignores components that are more than 30 days out of stock. This reduces the database size and makes the page usable as a mid-term workaround. However, as the catalog was unusable and I didn't have the data to just now ignore components that were 30 days out of stock, I decided to start with ignoring all components out of stock right now in order to make it usable again. On 25th November, the page will correctly show components in stock and components that have been seen in stock in the last 30 days.

It is possible that we will prolong this to 90 days in the future. I just don't have the data on component availability at the moment to make a proper decision.

closed 2022-10-09 by yaqwsx

UniversalChaos, 2022-10-09:
I do support the idea to only have the items listed that is on stock, we can't use anything that is not on stock.

asid61, 2022-10-09:
I am also having an issue where stock numbers don't show up. Furthermore, some parts show up as "in stock" even though they're not in stock. Still a great tool, just need to check stock manually.

yaqwsx, 2022-10-11:
@asid61: Could you give concrete examples of parts that have the wrong stock? Without examples, I cannot debug the problem.

asid61, 2022-10-16:
I did a sweep of some parts I was looking for, and I can't find any that are not in stock (although some will appear even if the "Specify Quantity" field is greater than the total number in stock. 

Here's a screenshot showing what I see when I look for parts. Note that no stocks shows. None of those parts actually have 10k stock.

[image]

zwn, 2023-02-22:
jlcpcb supports pre-ordering parts that are not in stock so ignoring parts that are not in stock makes the services less valuable. Can you add an explanation to the page explaining which parts should we expect to be found? Like stating explicitly that not all parts available from jlcpcb can be found using the tool and that this is the expected behavior.