yaqwsx/jlcparts #74: Stock disappeared

vtpgit, 2022-10-18:
One of the recent changes blanked out the stock number. Without the stock it is very difficult to make certain design decisions, as sometimes the stock is less than the number of required available components for a design, especially for rare and hard to find components which are naturally low stock. I understand this improves the speed and performance, however, the tradeoff is too big and not worth the improvement. Users need to switch all the time between the tool and the JLC web site to check the available stock.

Also, by hiding the out of stock components, certain design decisions are also very difficult to make, as the out of stock components cannot be searched for comparison and datasheet research.

Please bring back the stock, or at least give the user the ability to search with or without stock when initially querying/refreshing the database.

Cheers.

yaqwsx, 2022-10-18:
What do you mean by "stock number is gone"? Could you provide me with a concrete example? I don't understand what is the problem from your description.

At the moment, we don't show components that weren't in stock for the last 30 days, but I plan to extend this to 90 or 180 days. This seems reasonable to me as roughly 1/2 of the components listed by JLC PCB were never stocked.

yaqwsx, 2022-10-18:
Thanks for the report. It is fixed now.

closed 2022-10-18 by yaqwsx