yaqwsx/jlcparts #78: Update component database fail

ElliotCui97, 2022-11-26:
look like update component database failed 5 days
[image]

yaqwsx, 2022-11-26:
Yes, I know. JLC PCB component table is out-of-service at the moment. This was reported, and there is nothing I can do about it. You can urge JLC PCB to fix their services.

andylinpersonal, 2022-11-28:
[image]

They don't provide the link to the database now :(

andylinpersonal, 2022-11-28:
[image]

They don't provide the link to the database now 👎🏻

yaqwsx, 2022-12-06:
I tried discussing with JLC PCB and finding a compromise. But without success. Details in https://blog.honzamrazek.cz/2022/12/jlcparts-falling-down-to-ashes/.

PhilCS, 2023-01-01:
JLC is working on an API for KiCad plugin devs, maybe your search engine can be adapted to it once available: https://github.com/Bouni/kicad-jlcpcb-tools/issues/241#issuecomment-1367159022

yaqwsx, 2023-01-01:
I am in close touch with JLC, and I initiated their interested in developing API. However, at the moment it doesn't do what we need. I hope we can get the catalog back working late in January or early in February.

yaqwsx, 2023-01-11:
The service should be back operational via [3da7afa](https://github.com/yaqwsx/jlcparts/commit/3da7afa003f11d1251286e85e4c7b3d486efd722). We still need to implement #37 and we miss minimal order quantity and attrition info, as JLC PCB no longer provides them, though. But better than nothing.

closed 2023-01-11 by yaqwsx

PhilCS, 2023-01-11:
Cheers mate!