yaqwsx/jlcparts #85: Filter Idea

immortal-sniper1, 2023-03-09:
So I was thinking that maybe adding a check box for extended or basic parts would be very useful. since that way we can narrow down the search when we have some wiggle room, like E24 resistors seem all to be basic but I don't remember all the values that well but if I filter for basic parts I could narrow it down a lot, also JLC has some odd categories , there are basic leds in like 3-4 places and finding those gems would be that much easier with a basic check box.

yaqwsx, 2023-03-09:
The first parametry in property filter isn't enough?

[image]

immortal-sniper1, 2023-03-09:
i was blind and i did not see it, well in a way i was strugling a bit but i found it now THX

closed 2023-03-09 by immortal-sniper1