yaqwsx/jlcparts #86: Propertiy names should be cAsE inDEpenDeNT

ADonut, 2023-03-09:
'package' and 'Package' are different properties. This clutters the UI. Property names should be normalized and merged.

FWIW, capitalizing the first word only, 'Number of channels', seems to use less pixels than capitalizing every word, 'Number Of Channels'.

This will backfire on property names that need a particular capitalization, 'MHz' -> 'Mhz', 'MOQ' -> 'Moq'

DeflateAwning, 2023-11-06:
[In the codebase](https://github.com/yaqwsx/jlcparts/blob/d81cf93d5ef1fa9d707f2c5ecef5a66a4c82a7e3/jlcparts/partLib.py#L22), there's a funny comment that cautions against normalizing category names.

```python
def normalizeCategoryName(catname):
    return catname
    # If you want to normalize category names, don't; you will break LCSC links!
    # return catname.replace("?", "")
```

I still want to take a stab at this, but it may require some semi-major reworking as it appears that the category names are strongly linked to some URLs or similar.

yaqwsx, 2023-11-06:
Coincidently, I implemented 9402f0747e5611e4a70380abaa5a4ea9557c0163 yesterday, so the problem should be gone.

@DeflateAwning: The solution is straightforward; a category should also have a parameter for URL prefix. However, given how messy the data from JLC PCB are, I don't see a huge benefit in simplifying five or six category names. But feel free to open a PR for that.

The original issue is resolved, closing.