yaqwsx/jlcparts #91: weird component selection issue

raspico, 2023-07-17:
I noticed that component selection differs slightly vs the results on the JLCPCB site. I updated the database, and for instance the search for PY32 returns 19 results on https://jlcpcb.com/parts/componentSearch?searchTxt=ch32v but 17 results in https://yaqwsx.github.io/jlcparts/#/ and similarly for other search terms. Of course the website database is dynamically changing but I think that somehow the searches in your app are under-reporting on searches.

raspico, 2023-07-19:
That's a great idea but I don't think this is the explanation here. I searched for PY32; the LCSC website gives 25 hits, of which 2 are dev kits, so 23 valid parts. The database has only 19 hits; what's missing is PY32F003F18U6TR	PY32F030F38P6TU	PY32F071C1BU6TR	PY32F071R1BU6TR	
They all have availability in 3-4 digits. 

AHA! I just noticed something. I've been searching on LCSC website. The same search on the JLCPCB site returns 19 items, and they match your results. So, somehow LCSC has slightly different data (?). 

The bottom line is that this issue is a false-positive and can be closed, but I thought I'd mention the LCSC-JLCPCB difference because it may matter to someone.

BTW, I had a copy of the database from February lying around, so I checked and as you say, there's a million parts that had no stock both then and now, about 48k parts that went from no-stock to some-stock, and 32k parts that apparently aren't very popular because their stock didn't change :).

yaqwsx, 2023-07-21:
> AHA! I just noticed something. I've been searching on LCSC website. The same search on the JLCPCB site returns 19 items, and they match your results. So, somehow LCSC has slightly different data (?).

Yes, JLC PCB offers only a subset of components compared to LCSC for their assembly service.

closed 2023-07-21 by yaqwsx