yaqwsx/jlcparts #93: question: cache.sqlite3

ole00, 2023-09-01:
Hi,
I like the cache.sqlite3 file I downloaded from github.io (linked in your readme.md). It seems to be a bit out of date now (the timestamps seem to be from May 2023). How do I generate new cache file?  Or is it generated when the web page collects  the data? If so, where is the cache file located? Thank you.

yaqwsx, 2023-09-01:
Where did you download that file? The file was last updated 14 hours ago.

ole00, 2023-09-01:
From the link in readme.md :
wget https://yaqwsx.github.io/jlcparts/data/cache.zip 
.. and other zip parts: *.z01, *.z02 etc.

The time stamps in that file looks to be: 1682999876 (Tuesday, 2 May 2023 03:57:56).  I assume it is the timestamp of "last_update INTEGER NOT NULL" field.

yaqwsx, 2023-09-01:
The file should be updated in this run: https://github.com/yaqwsx/jlcparts/actions/runs/6044971793

ole00, 2023-09-01:
I see. The component_table artefact is quite helpful as well - thanks for that link. I'll use it instead of the caches.sqlite3.

ole00, 2023-09-01:
Closing the issue. Thanks for your help.

closed 2023-09-01 by ole00

ole00, 2023-09-02:
FYI, I've created a KiCad BOM exporter that selects basic parts from the LCSC database and assigns them to Resistors and Capacitors. This can be helpful if the KiCad project has plenty of standard value resistors and capacitors. The BOM exporter uses filtered data from the parts.csv file as produced by this repo.
The project sources and tools are hosted here: https://github.com/ole00/kicad_bom_jlcpcb_auto