yaqwsx/jlcparts #95: Preferred extended parts - free assembly

maksz42, 2023-09-24:
I was upset because lmv321 became an extended part but then I saw this in jlcpcb's browser :
[image]
Looks like now there is a whole subset of extended parts that they don't charge a fee to assemble?
Here is a list of these:
https://jlcpcb.com/parts/basic_parts
[image]
Would be great to have this in jlcparts :+1:

yaqwsx, 2023-10-26:
At the moment, is seems that JLCPCB doesn't provide this flag via their API.

brianbienvenu, 2024-01-04:
I've just found this too. Potentially a baked-in list of the Basic/EP item codes would suffice. My understanding is that these shouldn't change very often, to provide design stability. (That said, sometimes they do - C2293 and C2296 were previously Basic and are now Extended.)

maksz42, 2024-01-05:
Looks like all preferred parts can be obtained with a single request (additional requests may be needed to get cookies, tokens, etc)
There are currently 300 preferred parts and more and more are getting moved from basic to preferred category.
We can request all preferred parts using "pageSize" field
[image]
And the response contains a list of all preferred parts.
[image]
So an additional step could be added during the database build process to mark those parts as preferred.
@yaqwsx What do you think?

yaqwsx, 2024-01-06:
@maksz42: This looks good! But I prefer if you post it as text instructions to obtain the data. It is not particularly easy to deduce it from your screenshots.

Implementation-wise: I suggest adding a new component table column in the cache DB `preferred`. We add a new update step where we remove all preferred flags and set it only for the newly fetched components. When we build data tables from the cache DB, we will infer whether components are basic, preferred, or extended.

maksz42, 2024-01-06:
@yaqwsx I opened a PR. Since it's only a single request and I can't test it on cache DB, I added it to `buildtables`. So you'll need to change that if you want to keep this task offline.
I just took the API from https://jlcpcb.com/parts/basic_parts and removed headers and fields that seem like they don't matter. First we need to obtain xsrf token and then query jlcpcb for preferred parts.

yaqwsx, 2024-01-07:
Implemented in bb8f3311.

closed 2024-01-07 by yaqwsx