yaqwsx/jlcparts #96: Docker - YAML file

CzarnyLewis, 2023-09-25:
I have been trying to run JLCParts in Docker for several hours. Unfortunately, I don't know what the YAML file should look like. Does anyone know:
- what ports to map?
- what path to map?
- what environment to set?
Thank you

yaqwsx, 2023-09-25:
What are you trying to achieve? The CI of this project builds a static website. There is no backend. Just plain-old static website that is enough to just serve via any HTTP server.

CzarnyLewis, 2023-09-25:
I see there is a Dockerfile, so I thought it was cool to have a actual library of electronic components in a container.

yaqwsx, 2023-09-26:
That was dockerfile specifying the run environment for Travis CI we no longer use (I removed it as it is no longer relevant).

The build process is described in https://github.com/yaqwsx/jlcparts/blob/master/.github/workflows/update_components.yaml

Since running this inside Docker makes no sense (there is no backend to run), I close this issue.

closed 2023-09-26 by yaqwsx