yaqwsx/jlcparts #110: Filter Basic and Extended parts

besi, 2024-02-07:
When ordering with JLCPCB it's an financial advantage to use more basic parts.
https://jlcpcb.com/parts/basic_parts

Did you consider allowing people to filter those basic parts?

yaqwsx, 2024-02-07:
The filter is there since the project beginning:

[image]

Or did I misunderstand your feature request?

besi, 2024-02-07:
Very responsive. 
Thanks a lot.

closed 2024-02-07 by besi

besi, 2024-02-07:
I was looking for all basic parts and did not understand that the filters only appear once there are some results being fetched. My bad.

BTW: There might be a small bug on your screenshot with the `[object Object]` in the list of options.