yaqwsx/jlcparts #112: make a woo-commerce plugin

danyhm, 2024-02-15:
Hello,

Can you make this project into a woo-commerce plugin?

yaqwsx, 2024-02-15:
I am not familiar with woo-commerce. Based on quick search, I don't see how this should be turned into a woo-commerce plugin. I need to you elaborate on your request. Be more precise, provide examples and also provide a motivation what would we gain, what are the use cases.

danyhm, 2024-02-16:
Ok, I understand that LCSC part numbers and JLC SMT part numbers are the same. If someone could make a plugin out of it and sell it as a woo-commerce plug-in, many online electronic shops could implement it on their website and attract customers. Act as a 3rd party between the customer and LCSC. you'll be selling the plugin of course.
I've seen some online shops with their private implementation. they're not English, but you'll understand if you take a look.
https://digighate.com/
https://lionelectronic.ir/