yaqwsx/jlcparts #122: Test site for new features

dougy83, 2024-02-24:
I deployed a site which can be used for testing some new features, without taking out the original site with bad automerges and my bugs. This deployment is a side-effect of me needing to check if a workflow change I submitted actually worked (it initially didn't).

Site is https://dougy83.github.io/jlcparts/

Changes:
#114 new fast db update & search. Relates to/fixes issues #37, #45, #48, #109
#115 component count matching property. Relates to/fixes issues #76
#116 download results as CSV. Relates to/fixes issues #34
#117 filter by favourite components. Relates to/fixes issues #10
#120 (de)select all subcategories of the same name. Relates to/fixes issues #119
#121 search term negation. Relates to/fixes issues #103

The database file is updated after copying the .json.gz and .stock.json from the original site a couple of hours after they are created, as the original site has already done the hard work.

@yaqwsx if this post is out of line, please delete it (or I can).

yaqwsx, 2024-02-24:
It is completely fine! I was about to suggest submitting your PRs against https://github.com/yaqwsx/jlcparts-dev. Which is a development playground. Unfortunately, I cannot share API keys with you and this would be a way for you to test. However, the solution you came up with is also working and probably easier.

dougy83, 2024-02-24:
Excellent. Cheers.