yaqwsx/jlcparts #126: FR: Ranged attribute selection via min/max filter

dougy83, 2024-04-12:
For attributes having a range of value, e.g. 
[image]

It can be a bit of work to select e.g. all parts with a maximum input voltage >= 40V, as you'd need to select {"2.5V\~40V", "5V\~40V", "16V\~44V", "16V\~65V", etc.}

Items might be more easily selected if ranged attributes could be filtered by a _max-lower-value_ and a _min-upper-value_, which would allow selecting the matching attribute values from the multiselect box. For the example above, this would just require setting _min-upper-value = 40_

This change can be front-end-only, as ranges can generally be detected by their format.

dougy83, 2024-04-13:
Implemented in #127 
[image]

420pootang69, 2025-07-13:
Is this not available on the main site, or am I missing something?

[image]

dougy83, 2025-07-13:
The related PR hasn't been merged into the main site. In the meantime, you can use the new features in the test site linked in #122