yaqwsx/jlcparts #130: feature request: download ecad models

KJ7LNW, 2024-04-26:
It would be great if jlcparts can provide a link to download easyeda models.

You are half way there with EasyEDAFootprintScraper, though I know it expects python, so not sure how that could work in a JS-based UI.   Still, if you can come up with a way to "click-and-download" ecad models from your jlcparts engine, that would be awesome.

Cheers, and thanks for all the great work on these projects.

-Eric

KJ7LNW, 2024-04-26:
I just realized that EasyEDAFootprintScraper is now replaced with `easyeda2kicad`

Anyway, I'm not sure what the capabilities for `github.io` are for backend calls, but easyeda2kicad is pretty easy.  Something like this might work:
```
pip install easyeda2kicad

# pseudo-shell-code: replace sanitize_get_string($_GET["LCSC_ID"]) with something to prevent shell injection, and then return a .zip of the content:
easyeda2kicad --3d --debug --full --lcsc_id=sanitize_get_string($_GET["LCSC_ID"])
```

KJ7LNW, 2024-04-27:
Here is a POC: 
- https://eda.ewheeler.org/easyeda2kicad.fcgi?lcsc=C2040

This is the code, in Perl:

```perl
#!/usr/bin/perl

use strict;
use warnings;
use CGI::Fast;
use File::Temp qw/tempdir/;
use File::Copy qw/move/;
use Cwd qw(cwd getcwd);
use Data::Dumper;

$ENV{PATH}="$ENV{PATH}:/bin:/usr/bin:/home/eda_ewheeler_org/.local/bin";

BEGIN {
	open(STDERR, ">>/home/eda_ewheeler_org/easyeda2kicad.log");
}

my $easyeda2kicad = '/home/eda_ewheeler_org/.local/bin/easyeda2kicad';
my $cachedir = '/home/eda_ewheeler_org/easyeda-cache';
mkdir($cachedir);

#download_part('C2040'); exit;

while (my $q = CGI::Fast->new)
{
	my ($zip, $dir);

	my $lcsc = $q->param("lcsc"); 
	$lcsc =~ tr/A-Za-z0-9_.-//cd;
	if (!defined($lcsc) || !length($lcsc))
	{
		print "Content-type: text/html\r\n\r\n";

		print "Unknown LCSC_ID\n";
		next;
	}

	if ( -s "$cachedir/$lcsc.zip")
	{
		$zip = "$cachedir/$lcsc.zip";
	}
	else
	{
		($dir, $zip) = download_part($lcsc);

		if (-d $dir && -s $zip)
		{
			move($zip, "$cachedir/");

			# store the file as the model name, but symlink the LCSC_ID
			# for future reference:
			my $basename = $zip;
			$basename =~ s/^.*\///;
			symlink("$basename", "$cachedir/$lcsc.zip");

			rmdir($dir) if $dir;
			$zip = "$cachedir/$basename";
		}
	}

	if (defined $zip && -s $zip)
	{
		# If a symlink is accessed, use the linked filename:
		if (-l $zip)
		{
			$zip = readlink($zip);
			print $zip;
		}
		my $basename = $zip;
		$basename =~ s/^.*\///;

		print "Content-type: x-application/zip\r\n";
		print "Content-Disposition: attachment; filename=\"$basename\"\r\n";
		print "\r\n";

		catfile($zip);
	}
	else
	{
		print "Content-type: text/html\r\n\r\n";

		#print "Unknown LCSC_ID zip=$zip dir=$dir";
		print "Unknown LCSC_ID";
	}

}

sub download_part
{
	my $lcsc = shift;
	$lcsc =~ tr/A-Za-z0-9_.-//cd;

	my $dir = tempdir(CLEANUP => 1);

	popen($easyeda2kicad, '--full', '--output', "$dir/part", "--lcsc_id=$lcsc");

	if (! -s "$dir/part.kicad_sym")
	{
		return undef;
	}

	my $name = readsym("$dir/part.kicad_sym");

	if (!defined($name) || !length($name))
	{
		return undef;
	}

	#print "part: $name\n";

	mkdir("$dir/$name");
	mkdir("$dir/$name/KiCad");
	rename("$dir/part.kicad_sym",               "$dir/$name/KiCad/$name.kicad_sym");
	rename("$dir/part.pretty/$name.kicad_mod",  "$dir/$name/KiCad/$name.kicad_mod");
	rename("$dir/part.3dshapes/$name.step",     "$dir/$name/KiCad/$name.step");
	rename("$dir/part.3dshapes/$name.wrl",      "$dir/$name/KiCad/$name.wrl");
	rmdir("$dir/part.pretty");
	rmdir("$dir/part.3dshapes");

	my $cwd = getcwd();
	chdir($dir);
	popen(qw/zip -9 -r --move/, "$dir/$name.zip", "$name");
	popen(qw/unzip -l/, "$name.zip");
	chdir($cwd);

	return ($dir, "$dir/$name.zip");
}

sub popen
{
	my @args = @_;

	my $in;
	my $ret;
	if (!open($in, "-|")) {
		exec(@args);
	}

	while (defined(my $line = <$in>)) {
		$ret //= '';
		$ret .= $line;
	}
	close($in);
	#print "popen: $ret\n";

	return $ret;
}

sub readsym
{
	my $p = shift;

	open(my $in, $p) or die "$p: $!";

	while (defined(my $line = <$in>))
	{
		if ($line =~ /"part:([^"]+)"/)
		{
			close($in);
			return $1;
		}

	}
}

sub catfile
{
	my $fn = shift;

	#print "fn: $fn\n"; return;
	open(my $in, $fn) or print "$fn: $!";

	my $buf;
	while (sysread($in, $buf, 1024))
	{
		syswrite(\*STDOUT, $buf);
	}
}
```

KJ7LNW, 2024-07-12:
I just updated the perl LCSC sample downloader code above, it works great with the latest version of the Kicad [Imparter](https://github.com/Steffen-W/Import-LIB-KiCad-Plugin.git) plugin.  Imparter works with easyeda2kicad-generated archives after this commit:
https://github.com/Steffen-W/Import-LIB-KiCad-Plugin/commit/792ec670d0228af5ce16e8f8c28312383b55d0a4

sslupsky, 2025-08-11:
@KJ7LNW Thanks you for this work.  It is very useful.