yaqwsx/jlcparts #132: sqlite3.OperationalError: database or disk is full

Bouni, 2024-07-03:
Hi,

I just realized that you encounter the exact same issue as I do with my [KiCAD plugin](https://github.com/Bouni/kicad-jlcpcb-tools) as we use your cache.

https://github.com/yaqwsx/jlcparts/actions/runs/9771283483/job/26973813599#step:5:258

My guess is that the GH actions runners have disk that is now insufficient to store the database file ...

Do you have a plan on how to solve this?

I lack the time at the moment, but I would really like to build an API that serves JLC parts data, sanitized and usable.

Bouni, 2024-07-12:
I'll look into yor suggestion as soon as I have a bit of spare time 👍🏽

yaqwsx, 2024-07-23:
As a quick work-around, I tried to save some space on the CI machine. At the moment, it has been working just fine for 4 days now so the problem is solved - at least for now. I like the proposal of @dougy83, and I will look into it in the future.

dougy83, 2024-07-23:
@yaqwsx I only found fuse-zip to work for reading from the database, which it does very well. When I tried writing to it the zip file became corrupted.