yaqwsx/jlcparts #143: Parallelizeproduct searches

KJ7LNW, 2024-09-12:
it appears that javascript supports parallel workers:
 - https://stackoverflow.com/a/11871543/14055985

I am no javascript expert, but perhaps [jlcparts](https://github.com/yaqwsx/jlcparts) partition searches based on the number of CPUs (for some configurable field) to speed up the in-memory search?

dougy83, 2024-10-02:
Full text search takes <2seconds if you use the newer compressed database (not querying from IndexedDB). See [feature-test-site](https://dougy83.github.io/jlcparts/#/)

KJ7LNW, 2024-10-02:
That site is much faster! I was using https://yaqwsx.github.io/jlcparts/#/ .  Is "yaqwsx.github.io" it no longer maintained?

dougy83, 2024-10-03:
> Is "yaqwsx.github.io" it no longer maintained?

I believe it's still maintained, and if something breaks it gets fixed, but the owner is a bit busy to accept large changes. If I see a new issue I'll generally just fix it on my test site and send a PR to this repo.

The database on this site is updated daily, which I rely on to update mine.