yaqwsx/jlcparts #145: Please add option to parse/understand voltage/current range.

AndKe, 2024-10-30:
Selecting all regulators that can handle >54v is  quite a task

[image]

It would be very nice if one could search for ">50V"   or 1A , instead of manually selecting all the lines that fall within.

Thank you for a great tool.

dougy83, 2024-11-01:
This is addressed in #126 
It is implemented in my test fork: [link](https://dougy83.github.io/jlcparts)

AndKe, 2024-11-01:
please see: 
[image]

Why dont I see the range tool for - for example current ?  (looking for 2.2uH  3-5A)

dougy83, 2024-11-01:
Range tool is only for attributes specified as ranges, e.g. 5V~24V, etc. The currents in your screenshot are single values, and sorted, so you can multi select from a start value to an end value using the shift key and clicking, as normal.

closed 2024-11-01 by AndKe