yaqwsx/jlcparts #147: lcsc.com/rest API Rate Limit Exceeded in GitHub Action Since 27th of December

CDFER, 2024-12-31:
### Broken GitHub Action: Update Component Database
The GitHub Action has been failing since [#1599](https://github.com/yaqwsx/jlcparts/actions/runs/12511268213) due to an API request rate limit exceeded error.

**Error Message**
```python
jlcparts.partLib.FetchError: 200: {"success":false,"code":437,"message":"API request rate limit exceeded.Retry after 1 minute.","result":null}
```
**To Reproduce**

1. Run the GitHub Action
2. The action will fail with the above error message after 100 requests (1.33%)

**Fix**
I have written a very basic retry mechanism to getLcscExtraNew() in partLib.py with a 60s delay to try and avoid exceeding the API request rate limit but I don't have the api key so I can't test wether this code fixes the issue

CDFER, 2024-12-31:
https://github.com/yaqwsx/jlcparts/compare/master...CDFER:jlcparts:master

yaqwsx, 2025-01-02:
I merged your changes. However, it seems that the API limit was lowered significantly. I will try to contact JLC PCB and find a solution.

CDFER, 2025-01-05:
Thanks for that it seems to be working again now :)

I have noticed that it still gets a rate limit warning at around 80% for around 40 component requests but each request succeeds after waiting for 1min-> see line 1044 here: https://github.com/yaqwsx/jlcparts/actions/runs/12607055051/job/35137990130

Anyway thanks for all you work on this repo hope you have a good holiday :)

closed 2025-01-05 by CDFER