yaqwsx/jlcparts #149: Where does the cached parts file come from? Would be great if it was a 7zip archive.

chmorgan, 2025-04-09:
Hello. I work on https://github.com/Bouni/kicad-jlcpcb-tools. I was wondering where the cached database comes from. I'm looking for a pure python way to uncompress the database. This *could* work if 7zip was being used, but nothing python native seems to support multi-part zip files, so you end up having to shell out to run an externally installed tool.

7zip is also much more effective at compressing the database, the same database compressed with zip is ~1GB, compressed with 7zip is ~500MB.

yaqwsx, 2025-04-10:
The cache DB contains parts definitions cached from the LCSC API.  The 50 MB part limit comes from the limitation of Github pages. There is nothing I can do about it.

chmorgan, 2025-04-10:
How do you get it from the LCSC API if you don't mind me asking?

The 50MB parts seems like a reasonable thing to do, it's the zip multi-part vs. 7zip multipart that makes it python incompatible, no one has put together a multi-part zip module for python yet and its not likely to be me. heh.

yaqwsx, 2025-04-11:
I guess I can't have a better and more accurate answer than - these are the entry points to the update routine: https://github.com/yaqwsx/jlcparts/blob/5fa91e1562a69d0abdd0246d11d20f07c3cb28ca/.github/workflows/update_components.yaml#L61-L66. The rest of the code is in this repository.

Regarding switching to 7Z - you are not the only one using this data source, and I don't want to introduce a breaking change. Also, I am hesitant to provide two versions of the same data as I am already quite stretching what is GitHub intended for. That's why I propose - feel free to repack it and rehost it.

closed 2025-07-05 by chmorgan